lake-rs/lakers #67: EAD processing mechanics

malishav, 2023-05-02:
EAD - external authorization data provided by EDHOC.

geonnave, 2023-05-12:
I think we could organize the callbacks within implementations of "handlers", which would hold some state for the EAD processing and implement the needed callbacks. The callbacks should be enforced by a contract, which can be achieved by Rust's traits.This way, for example, we would have the application as follows:

```rust
let edhoc_zeroconf_handler = EdhocZeroConfHandler::new(LOC_W, I, G_W, ID_CRED_I, CRED_I, ID_CRED_W, CRED_W);
let mut initiator = EdhocInitiator::new(EdhocState::default(), I, G_R, ID_CRED_I, CRED_I, ID_CRED_R, CRED_R);
initiator.register_ead_handler(edhoc_zeroconf_handler);
```

Then in i_prepare_message_1 we would add something like:

```rust
for ead in ead_handlers.iter() {
    message_1.append(ead.get_label());
    message_1.append(ead.i_prepare_ead_1());
}
```

Note that get_label and i_prepare_ead_1 would be part of the contract/trait  that needs to be implemented by all handlers.I have prepared a gist with a little bit more of the coding details: https://gist.github.com/geonnave/826da228642f8c024f80e46796a06fec

malishav, 2023-05-12:
I think that this is a good approach. hacspec v1 does not support traits so we never used them in the code, but v2 does, so we can get fancy. I am just thinking whether it would be better to have 2 separate traits: one for EdhocEADInitiator and one for EdhocEADResponder. That way, we can cleanly split who implements what, which might have benefits in terms of the code size. What do you think?

In terms of the gist, one comment on https://gist.github.com/geonnave/826da228642f8c024f80e46796a06fec#file-proposal_edhoc_ead-rs-L34:
- You will need to implement the logic of critical (negative) vs non-critical (positive) EAD item processing that is specified in Section 3.8: https://www.ietf.org/archive/id/draft-ietf-lake-edhoc-19.html#name-external-authorization-data

geonnave, 2023-05-12:
Great, I will proceed with it then.

geonnave, 2023-05-16:
The Rust type system imposes some restrictions on the implementation.

The main problem is how to store the list of EAD handlers. From that, other implications also follow.

Basically, we have two options:
1. use trait objects (requires heap allocation):
  - allows having a list of "things that implement a given trait"
    - this will also provides dynamic dispatch
    - the "thing" (a `struct`) will live in the heap
  - allows defining `struct`s AND `impl`s in the application code
2. use fixed fields (stack only):
  - need to declare all possible EAD item types (for example, add ead_* fields to the `State` struct)
  - the `struct`s need to be defined in edhoc-rs, as well as its (non-`trait`) implementations
  - must pass each callback manually (`traits` cannot be implemented for `structs` defined in another crate)