lake-rs/lakers #76: CoAP server's behavior on repeating EDHOC Message 3

actyp, 2023-06-04:
We are developing a model learning and testing tool, called EDHOC-Fuzzer, for EDHOC clients and servers, and we have used the implementations found in `examples/coap` in order to learn their models and analyze their behavior.

We have checked the version of 09aa2a8 and lately the release version of a3abcbb.

The client's model looks good in both versions, but in the server's model some problems have been observed.

In order to keep issues dedicated to a single behavior, this issue refers to the server's learned model of 09aa2a8 and its behavior on **repeating EDHOC Message 3**.

### 09aa2a8 Server Test: Two repetitive EDHOC Message 3 after successful EDHOC exchange
```
    CLIENT            SERVER
(EDHOC-Fuzzer)    (Under Testing)
      |------  M1  ----->| * EDHOC Message 1 from EDHOC-Fuzzer
      |<-----  M2  ------| * EDHOC Message 2 from server
      |------  M3  ----->| * First EDHOC Message 3 from EDHOC-Fuzzer
      |<----- EMPc ------| * The EDHOC exchange is completed
      |------  M3  ----->| * Second EDHOC Message 3 from EDHOC-Fuzzer 
      |<----- EMPc ------| * The server responds to the second M3 similar to the first M3
      |------  M3  ----->| * Third EDHOC Message 3 from EDHOC-Fuzzer
      |<----- EMPc ------| * The server responds to the third M3 similar to the first M3
```

#### Messages used

 * M1 (EDHOC Message 1)
 * M2 (EDHOC Message 2)
 * M3 (EDHOC Message 3)
 * EMPc (Empty CoAP message, which means it has no payload)

#### Accompanying Files

We can also provide you with the server's log and a wireshark capture of this interaction (either in this issue or via email).

#### Observed Behavior

The server responds to following M3 messages (which are the same as the first M3 message in terms of payload) in the same way as it responds to the first M3 message, which terminates successfully the EDHOC exchange.

### Some Questions

 * Can you confirm this behavior?
 * Were you aware of it?
 * Is it intentional or an oversight?

malishav, 2023-06-05:
Thanks for reporting this. We are working on a patch and will respond with more details ASAP.

geonnave, 2023-06-06:
Hi @actyp, thanks for the report.

We confirm the behaviour, which was not intentional.

The problem though was not in the library state machine, but in the server (example) code, specifically how it manages instances of the `responder` in the `edhoc_connections` list.

We have provided a fix in #78.