lake-rs/lakers #111: Features not worked for build command in examples\edhoc-rs-no_std on system Windows

ysong02, 2023-10-05:
Failed to build for crypto-psa and crypto-cryptocell310 in Windows environment.

malishav, 2023-10-05:
@ysong02 could you update hte issue with the exact output of the build command that you get? thanks!

ysong02, 2023-10-05:
[image]
for software

ysong02, 2023-10-05:
[image]
for hardware

ysong02, 2023-10-06:
Solved build problem with cryptocell310 cryptographic backend by installing LLVM on Windows machine, and set LIBCLANG_PATH as an environment variable pointing to the bin directory of the LLVM install.

geonnave, 2023-10-09:
So is this fixed now? If yes feel free to close it @ysong02.

ysong02, 2023-10-09:
Still failed to build by psa crypto on Windows, the hardware one worked.