lake-rs/lakers #152: License ambiguity

chrysn, 2023-11-20:
The `license = "..."` fields of the crates contain the identifier "BSD", which is not a recognized SPDX identifier ([even though that's the list into which that key maps](https://doc.rust-lang.org/cargo/reference/manifest.html#the-license-and-license-file-fields)).

From the LICENSE.md file, my best guess is that you intend to put "BSD-3-Clause" in there. When addressed after #146, this can easily be addressed widely by placing a license term in the workspace Cargo.toml, and linking it from the individual crates using `license.workspace = true`.

malishav, 2023-11-21:
Yes, confirming BSD-3-Clause is the right license.

geonnave, 2023-12-04:
Solved in #146

closed 2023-12-04 by geonnave