lake-rs/lakers #215: Remove hacspec crypto backend

geonnave, 2024-02-12:
- it is not needed anymore -- since the addition of the `rustcrypto` backend, which now serves as the default backend for non-embedded environments
- and crucially, it has been [causing CI to fail](https://github.com/openwsn-berkeley/lakers/actions/runs/7831541740/job/21368143161), with a hard to diagnose cause (it was pulled from a stale fork)

chrysn, 2024-02-12:
Are there not potential proofs that depend on its kind of expression?

geonnave, 2024-03-15:
@chrysn sorry for the late response. I think it could have, but I just re-tested and the build still fails. Last time I checked with the hax team, they could not identify the source of the issue. Finally, given that the proof is currently being done using the rustcrypto backend, I think it makes sense to proceed with the removal.

closed 2024-03-18 by geonnave