lake-rs/lakers #234: Error parsing message 2, likely a pointer issue in the FFI

geonnave, 2024-03-06:
When integrating the latest version of `lakers-c` in the DotBot, `initiator_parse_message_2` always fails with `ParsingError`.
This behavior was also observed my a native target, although only happening intermittently.
I think the same issue was found when @malishav used `lakers-c` with the `AIOT Play` platform, although in that case initializing all to zero solved the issue.

Specifically, the error happens in `decode_plaintext_2`, where the `plaintext_2` parameter seems to be overwritten by seemingly random bytes.

I suspect of an issue with pointers in the FFI boundary, but so far I could not isolate the exact source.

The plan of action is to research and implement guidelines for safer Rust-to-C FFIs, and re-test.

chrysn, 2024-03-07:
Are any of the tests running with ASAN or similar tools?

geonnave, 2024-03-07:
ASAN is new to me, I am using it to investigate, thank you for the pointer.

chrysn, 2024-03-07:
[image]

chrysn, 2024-03-07:
It may also be UBSAN, they're best used both together.

geonnave, 2024-03-13:
In the end, the bug was not in the FFI. Actual source is #242, and it was fixed by #243.

PR #240 (2024-03-13): Towards safer code in the FFI
The goal is to fix #234.
closed 2024-03-13 by geonnave