lake-rs/lakers #268: Python tarball references non-shipped example

chrysn, 2024-05-15:
Building the Python package from source [for pyodide](https://github.com/pyodide/pyodide/pull/4763) shows this error:

```
error: rustup is not installed at '/root/.cargo'                                
info: component 'rust-std' for target 'wasm32-unknown-emscripten' is up to date 
Running build script in                                                         
/root/repo/packages/lakers-python/build/lakers-python-0.3.0                     
error: package                                                                  
`/root/repo/packages/lakers-python/build/lakers-python-0.3.0/examples/coap` is  
listed in workspace’s default-members but is not a member.                      
ERROR: build script failed                                                      
    cargo update -p proc-macro2 --precise 1.0.60                                
                                                                                
[2024-05-14 12:11:17] Failed building package lakers-python in 0.8 seconds.     
```

I don't understand fully what is happening yet, but I could confirm that downloading the [tarball from pypi](https://files.pythonhosted.org/packages/29/93/5d70b035f987a48dd854c1afc21025fb8446aae4d43c685b68691175623c/lakers_python-0.3.0.tar.gz) and unpacking it gives a checkout without examples but that mentions examples/coap in its Cargo.toml default-members.

Not sure how all this is generated; @geonnave, do you have any ideas?

chrysn, 2024-05-15:
Note that `maturin build` appears to work just fine in the checked out source tar ball, just `cargo check` shows this error.

chrysn, 2024-05-15:
The immediate impact on the pyodide PR could be remedied by removing a needless `cargo update`, but shipping a package in which most cargo commands throw an error is still a weird situation that should be remedied.

geonnave, 2024-05-16:
IIRC pyproject has a flag for including specific folders, will take a look.

geonnave, 2024-05-17:
Hopefully solved by #275. Will publish new python package version soon.

geonnave, 2024-05-17:
Should be solved in 0.3.1 -- https://pypi.org/project/lakers-python/

closed 2024-05-27 by geonnave