lake-rs/lakers #282: Python package: export type stubs

chrysn, 2024-05-23:
Right now when you run mypy on a project with lakers, mypy doesn't find the stubs:

```
$ python3 -m venv .venv
$ source .venv/bin/activate
$ pip install lakers
$ echo "import lakers" > some.py
$ mypy --install-types .
test.py:1: error: Cannot find implementation or library stub for module named "lakers"  [import-not-found]
test.py:1: note: See https://mypy.readthedocs.io/en/stable/running_mypy.html#missing-imports
Found 1 error in 1 file (checked 1 source file)
```

I'd expect that typing information is available somewhere to pyo3, maybe we just have to enable exporting it. I'm not sure how it is done, and have a similar issue open at https://github.com/chrysn/cbor-diag-py/issues/2

Until this is resolved, users such as aiocoap will use mypy overrides to disable type checking around lakers (and link to this issue, which is one of the major reasons I'm creating an issue for it ;-) ).

geonnave, 2024-05-27:
Apparently automatic stub file generation is not (yet) available on pyo3. There is some ongoing work here: https://github.com/PyO3/pyo3/issues/2454

chrysn, 2024-05-27:
Thanks for the pointer. Given they're on it, AFAIK we have no pertinent insights to offer, and we have no hurry that'd warrant a workaround, I suggest we keep this open as blocked by <https://github.com/PyO3/pyo3/issues/2454> and revisit when that is done.