lake-rs/lakers #288: lakers-python: Build also for pypy

chrysn, 2024-06-22:
The wheels uploaded for lakers-python do not contain wheels for pypy; this means that pypy hosts have to have a local Rust installation to install it.

It's not a big deal, but I think it should be tracked (it may be related that it's a workaround I put into the aiocoap build scripts that increases CI times, and I am in the habit of associating workarounds to a link to what they are working around ;-) ).

I have had a brief look at my own maturin "reference" project, cbor-diag, which has pypy builds, and I don't even remember ever having done anything to make that work, but I'll have another look to see whether this can be fixed easily.

geonnave, 2024-06-23:
I think I initially [restricted the build](https://github.com/openwsn-berkeley/lakers/blob/main/.github/workflows/python-wheels.yml) to (C)Python 3.{10, 11, 12} to based on [this table](https://devguide.python.org/versions/), trying to balance build time and potential demand from users.

Which version of pypy should I add to the build? [Based on this](https://docs.github.com/en/actions/automating-builds-and-tests/building-and-testing-python#using-multiple-python-versions), I am not sure how is the support of GitHub Actions to newer Python versions on pypy (btw I am not familiar with pypy).

chrysn, 2024-06-23:
10, 11, 12 is a good set at the moment. (Adding 3.13 could be nice soon, I may give that a test run).

The one thing I can tell you about users is that running on pypy3.10 is what I do on aiocoap's CI (which is where I noticed this). Personally I build on pypy as a check against cpython-isms (which admittedly lakers won't need, taking from pyo3, but the whl builds are for the downstreams).

closed 2024-08-19 by geonnave

chrysn, 2025-02-06:
Just as a remark to justify why a workaround in aiocoap's CI still references this issue:

We build for `pypy310_pp73-manylinux_2_34_x86_64` (and some other platforms); the current pypy:3 image only supports up to `pp310-pypy310_pp73-manylinux_2_31_x86_64`.

I don't see anything actionable here so the issue stays closed: I don't think we'll meddle with the defaults); on the long run, and as all involved pieces of software are updated, things will converge.

PR #291 (2024-08-19): More Python targets: 3.13, pypy3.10
On top of #290, this adds pypy3.10 and Python 3.13 to the list of build wheels.

Closes: #288

Feel free to disregard this as "too much CI time" if you feel it doesn't carry its own weight – that'd be a fine policy which I can then reference in the aiocoap CI setup as the reason why rust needs to be installed on those.