lake-rs/lakers #289: lakers-python: Fails to build from dist tarball (at least on Python 3.13)

chrysn, 2024-06-23:
Where no wheels are available (on pypy as on #288, or on Python 3.13 because that's not released yet so doesn't get built by default I guess), builds fail related to the missing lakers-crypto-psa crate.

To reproduce:

```
$ docker run --rm -it docker.io/python:3.13-rc /bin/bash
$ curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh /dev/stdin -y
$ . "$HOME/.cargo/env"
$ pip install cbor_diag
[succeeds -- just to show that Rust based crates can be built]
$ pip install lakers-python
[...]
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [23 lines of output]
      error: failed to load manifest for workspace member `/tmp/pip-install-galefl6n/lakers-python_3efcab44afde4189bc4835002ee9b6c9/lakers-python`
      referenced by workspace at `/tmp/pip-install-galefl6n/lakers-python_3efcab44afde4189bc4835002ee9b6c9/Cargo.toml`
      
      Caused by:
        failed to load manifest for dependency `lakers`
      
      Caused by:
        failed to load manifest for dependency `lakers-crypto`
      
      Caused by:
        failed to load manifest for dependency `lakers-crypto-psa`
      
      Caused by:
        failed to read `/tmp/pip-install-galefl6n/lakers-python_3efcab44afde4189bc4835002ee9b6c9/crypto/lakers-crypto-psa/Cargo.toml`
      
      Caused by:
        No such file or directory (os error 2)
      💥 maturin failed
        Caused by: Cargo metadata failed. Does your crate compile with `cargo build`?
        Caused by: `cargo metadata` exited with an error:
      Error running maturin: Command '['maturin', 'pep517', 'write-dist-info', '--metadata-directory', '/tmp/pip-modern-metadata-8f1fdte4', '--interpreter', '/usr/local/bin/python']' returned non-zero exit status 1.
      Checking for Rust toolchain....
      Running `maturin pep517 write-dist-info --metadata-directory /tmp/pip-modern-metadata-8f1fdte4 --interpreter /usr/local/bin/python`
      [end of output]
[...]
```

(Note that while pip complains about a newer version being available, `pip install --upgrade pip` doesn't change anything about the actual result)

geonnave, 2024-06-23:
Weird error about psa. 

Maybe it could be related to this warning which has been hanging around for some time already, but I am not sure.

```
warning: /Users/geovane/Developer/inria/dev/lakers-FORK/crypto/Cargo.toml: `default-features` is ignored for lakers-crypto-psa, since `default-features` was not specified for `workspace.dependencies.lakers-crypto-psa`, this could become a hard error in the future
```

chrysn, 2024-06-23:
I don't think so, my assumption (currently testing the patch for it) is that it's like the earlier `examples/` issue, where the crate's presence in the manifest Cargo.toml means that we'll have to ship it even if we don't use it.

PR #290 (2024-06-23): Make sdist packages usable
Closes: #289
closed 2024-06-23 by geonnave