lake-rs/lakers #300: Widen usable C_x range

chrysn, 2024-08-04:
Currently we only support the numeric C_x values. This has caused trouble at some interops so far, especially when the peer randomly selects one, or the peer selects b'' before b'00'.

The data structure I propose to use is this:

```
struct Cx {
    full_encoded: Buffer<MAX_CX_LEN>,
    is_compact: bool,
}
```

with the validity constraint that:
* full_encoded is a valid CBOR item that is a byte string in canonical encoding (which, in the expected case that MAX_CX_LEN is <= 25, means that the first byte is always 0x40 masked 0xe0, and masked 0x1f is equal to the length - 1)
* is_compact is true iff full_encoded.len() == 2 and full_encoded[1..2] is a CBOR item that represents a small integer.

This can have a method to view the full version, to view the content (relevant to OSCORE), and to view the compact version (which is the full version if is_compact == false, else it is full_encoded[1..2]).

geonnave, 2024-08-05:
I acknowledge that this needs to be fixed.

geonnave, 2024-11-25:
Solved in #320

closed 2024-11-25 by geonnave