lake-rs/lakers #332: Release v0.8.0

geonnave, 2025-01-22:
Mainly include the addition of message_4 processing.

chrysn, 2025-01-22:
Can we finish the discussion around and then https://github.com/openwsn-berkeley/lakers/pull/331 before releasing this?

geonnave, 2025-01-22:
Absolutely!

chrysn, 2025-01-22:
I'll file one brief PR still: updating EAD labels to be 16bit. Nothing to fix how they're broken for labels > 24 -- that we can fix properly later, but up'ing the type is an API change.

chrysn, 2025-01-22:
Filed as https://github.com/openwsn-berkeley/lakers/pull/333, please consider for inclusion in 0.8.

chrysn, 2025-02-03:
All the things on my "please include this on 0.8" list are merged; while I have two more open, those are non-breaking and could just as well go in later (but if they're ready, they're nice to have).

PR #344 (2025-02-06): release: rust 0.8.0 and python 0.5.0
fixes #332
closed 2025-02-06 by geonnave