lake-rs/lakers #347: C error len is not set

geonnave, 2025-03-16:
The len field in EdhocMessageBuffer is not being set.

I isolated this to happen just after merge of PR #333.

Logs:

```
~/D/i/d/l/e/lakers-c-native ❯❯❯ make clean all LAKERS_EAD=authz && ./lakers_c_native                                                                                    ✘ 1 tags/v0.8.0~4 ✭ ◼
rm -f main.o lakers_c_native
gcc -Wall -I../../target/include -fsanitize=address,undefined,leak -DLAKERS_EAD_AUTHZ -ggdb -c main.c
gcc -Wall -I../../target/include -fsanitize=address,undefined,leak -DLAKERS_EAD_AUTHZ -ggdb -O0 -o lakers_c_native main.o -L../../target/x86_64-unknown-linux-gnu/release/ -llakers_c -lcoap-3
Calling lakers from C!
loading credentials.
creating edhoc initiator.
creating ead-authz device.
computing authz_secret.
computing ead_1.
0 bytes: 
Begin test: edhoc initiator.
0 bytes: 
sending msg1
sending coap message of size 0+1
1 bytes: F5
received coap response
1 bytes: F5
processing msg2
Error parse msg2: 6
```

geonnave, 2025-03-16:
Still confused about why it happens, `label` was correctly updated to `u16`

```
pub struct EADItemC {
    pub label: u16,
    pub is_critical: bool,
    pub value: EdhocMessageBuffer,
}
```

malishav, 2025-03-25:
Just bumping this up as it bit me today

chrysn, 2025-03-26:
To reproduce this, I went through the following steps:
* unset the global Cargo config target-dir (already for the C building, where I had to ignore several warnings)
* switch from -lcoap-3 to -lcoap-3-gnutls (working off libcoap develop branch, README is not clear on versions)

and because I didn't install libcoap system-wide:
* s/=/+=/ on the CFLAGS and LDFLAGS in Makefile
* set `CFLAGS="-I ~/git/libcoap/include"` in the environment
* explicitly add LDFLAGS in the TARGET rule
* run as CFLAGS="-I ${HOME}/git/libcoap/include" LDFLAGS="-L ${HOME}/git/libcoap/.libs/" make clean all LAKERS_EAD=authz && LD_LIBRARY_PATH=${HOME}/git/libcoap/.libs/ ./lakers_c_native

… but all that still only tries to run a CoAP server, expecting a peer that is not documented in the C test's README.

Is there any more minimal version of this?

malishav, 2025-03-27:
I have an example ready for Segger and nRF52840 that is complete but I think the `lakers-c-riot` example is more lean. Have you tried that one?

malishav, 2025-03-27:
Thanks for doing that, wasn't aware that this one is not updated. What do you get as the length of `message_1`?

chrysn, 2025-03-27:
I see EAD1 with a length of 0 and even message 1 has a length of 0 ... ?

chrysn, 2025-03-27:
0.7.2 gives me data, bisecting…

chrysn, 2025-03-27:
So where I am now (after bashing things long enough to get a debug build on the host) is that the C side thinks the EAD buffer is 320 bytes, while the Rust side thinks it is 192 bytes long. Reaching into the length that is assumed to be 192 byte after the start of the buffer when really 320 bytes of buffer have been assumed by the entity populating them just gives lengths 0, and the Rust function that populated them wrote into some wrong memory.

Still looking for *why* they disagree.

If we do C, we should run some examples in CI, and run with asan+ubsan!

closed 2025-03-27 by malishav